Mazamars312/Analogue_Pocket_Neogeo #3: Predefined folders

tkne, 2022-08-01:
There seems to be an issue with using the predefined folders which you provide. When placing the rom files into a game folder it ends up blank on the Pocket.
Using the a game‘s folder as is from the darksoft set and copy paste the game.json file from the predefined game folder into it seems to solve this issue. I could reproduce it and it happens every time. I don‘t know if this happens on a Windows PC but it does happen on macos in my case.

SegaSnatcher, 2022-08-01:
It's the JSON file in thise folders that is causing the issue.  Don't use those folders until he implementes auto-loading of all assets.

tkne, 2022-08-01:
Thanks for the heads up. True, I didn‘t try it without the game.json file. Thanks for clarifying.

closed 2022-08-01 by tkne