Mazamars312/Analogue_Pocket_Neogeo #4: Core Doesn't support Stereo version of XenoCrisis

Goldanas, 2022-08-01:
There are 2 versions of the XenoCrisis rom that you get when you purchase from BitMap Bureau:
- A Mono mixed version with everything in one 16 MB vrom
- A Stereo version with a 32 MB vrom (I believe two banked 16 MB roms, concatenated together)

The current version will throw a warning over this 32 MB vrom size, which is probably expected as its abnormal, but there does appear to be a way to implement for proper playback as it has been done via MiSTer.

The original file does have a special parameter to be included with the romset.xml listing for MiSTer, which is: vromb_offset="0x1000000"
The ROM can be officially purchased here:
https://shop.bitmapbureau.com/products/xeno-crisis-neo-geo-aes-mvs-rom-download

alexcom84, 2022-08-01:
Both versions are stereo, it just has higher quality samples. You’re confusing it with the Xenocrisis cart version with the mono audio bug that required a CPLD and M1/P1 rom update.

Goldanas, 2022-08-01:
I am not:
https://www.patreon.com/posts/missing-files-60446967

[image]

Mazamars312, 2022-08-01:
I will need to look into this as I only had the Neogeo made roms and I didnt expect larger VROMS of 16Mbyte and CROMs of over 64Mbytes. 

The Limits of the ram are as follows
CROM: 64Mbytes
VROM: 16Mbyte
PROM 12Mbyte
Z80 Program: 512Kbyte
SFIX: 256Kbytes (This will be increased to 512Kbyte soon)

I will add this as part of the Readme that the High Stereo version will not work at this moment.

Goldanas, 2022-08-01:
Yeah, it seems like a special case where you'd have to drop it into the sdram along with the small amount of crom this game has as a special exception. Thanks for taking a look.

closed 2022-08-02 by Mazamars312

dji85, 2022-09-19:
@Mazamars312 I tried the stereo version on the new core but am still seeing the same issue (Error in framework File ID [7] too large), thanks!

eobet, 2024-08-11:
Almost two years on, I'm seeing the same issue still unfortunately.