Mazamars312/Analogue_Pocket_Neogeo #5: Default Button Mapping Suggestion

Goldanas, 2022-08-01:
When the user remappable controls become a thing, this may not matter, but I'd recommend adjusting the default button layout to the recently released official SNK Neo Geo pads that came with the MVS Mini and Arcade Stick Pro:
[image]
This would look like this on the Pocket (image by terminator2k2 on Discord):
[image]

This is an optimal layout for many games. Many fighting games (certainly the most popular) have punches on A and C, and kicks on B and D. For games like Metal Slug and other platforms, action is on A, and jump on B (with a third option on C), and this also works well.

Mazamars312, 2022-08-02:
The Remapping will be part of the Analogue APF Json file soon. However I will change this for the next release. Adding to the Todo list.

RevQuixo, 2022-08-02:
[image]

Personally the official Neo Geo CD pad has a better layout in my opinion.

Wabbajackw, 2022-08-04:
RevQuixo is right, it's best optimized with the neogeo cd controller, I use it with my Mister for the most authentic experience and it's perfect

dji85, 2022-08-13:
Agree with @RevQuixo , that way you don't end up with a funky layout when used with the dock and an arcade stick too.

VideogameScrapbook, 2022-09-02:
Confirmed with Analogue OS v1.1 beta 2 that this works now with newly available Core Settings:
* Player Type 1 = SNES layout (default)
* Player Type 2 = MVS mini
* Player Type 3 = Neo Geo CD

JackWitherell, 2022-09-06:
changing `"default": "0"` didn't work for me so my workaround (not ideal) for forcing my controller choice in the meantime in `interact.json`:
[image]

VideogameScrapbook, 2022-09-06:
@JackWitherell, doing that doesn't work for me (or at least when setting min: 1 to select MVS mini layout). When entering the menu, it successfully shows the player type selected based on this modification, but the actual button layout is still the default until pressing A on the setting in the core menu. Then when starting a new game, have to do process again, so it seems better to leave min as 0.

JackWitherell, 2022-09-07:
Wow, you're entirely right. What a mess.

After messing around a bit more I've solved some of the problems with this.
I've finally gotten it working. Try this:

[image]

Mazamars312, 2022-09-26:
I hope the current setup in the interact works for now.

closed 2022-09-26 by Mazamars312