Mazamars312/Analogue_Pocket_Neogeo #12: Bios Unreadable

Llorean, 2022-08-14:
With the suggest BIOS, in some games the options are not readable.
[image]
When an option is toggled or changed, it's clear something is changing (vague collections of white pixels appear, not enough necessarily to tell what shape they represent) but it's more or less impossible to tell what I'm setting the values to.

Mazamars312, 2022-08-14:
Can you show which bios, game and game files you are using. as Metal slug 5 works for myself on the uni-bios_1_0.rom.

Then i can back look at this

Llorean, 2022-08-15:
I'm using the uni-bios_1_0.rom with the Darksoft set.

The md5sum of the .rom is 6293999bbc32e594aa0ae1da2113dc4d
I get the same results with Metal Slug 5 if I choose the normal version or the AES version.

Mazamars312, 2022-08-15:
Oh I see it now. Let me have a look. Might be a sfix issue that ill need to look at.

Mazamars312, 2022-08-15:
Confirmed this is the sfix I need to fixup. This will be in the next build I do. Please give me some time on this, but thank you for showing this.

Mazamars312, 2022-09-18:
This should be fixed now in 0.7.5 as I have also added the sfix.sfix from to the system. Please enjoy

closed 2022-09-18 by Mazamars312