Mazamars312/Analogue_Pocket_Neogeo #13: Link to JSON game files has gone

therourke, 2022-08-15:
The .zip containing all the game json files is not included in the new release or mentioned in the instructions. It is causing a few headaches for people.

urbanvanilla, 2022-08-15:
They are already in the release and in the right folder. It's actually less confusing isn't it?

therourke, 2022-08-15:
I have the new core setup and everything is working. But for some people who are having issues (talking on forums like reddit etc) these json files and their location is causing confusion.

I think it just needs clarifying that this has changed and what can be deleted from the previous setup.

dallowbg, 2022-08-15:
I don't see how it causes an issue. Even if you left the original json files alone, game loading still worked once you copied everything over.

JonAbrams, 2022-08-16:
The problem is the outdated tutorials I think.

Mazamars312, 2022-08-16:
It was in the readme under "installation guide" to run this as a clean installation from 0.5.0. 

If you do have issues loading files, do check the game JSON's and check that the game is in the right directory as well.

As to the locations of the files this can be mass changed using the json generator that Terminator2k2 made and from there you can use this to relocate where your games are.

closed 2022-08-17 by Mazamars312