Mazamars312/Analogue_Pocket_Neogeo #14: Neo Turf Masters - gaps in UI sprites

itsTrifle, 2022-08-15:
Some gaps in UI sprites can be seen in the player banner that appears at the start of the turn and around the shot power gauge - see the arrow at 50% and the low/nice/high labels.
Does not appear in the current MiSTer core.
Pocket 
[image]
Mister NeoGeo 20220403
[image]

Mazamars312, 2022-08-16:
Will look into this after moving the ram around in the core.

Mazamars312, 2022-09-18:
This has been partly fixed now... BUT I need someone to fully check in 0.7.5 please

itsTrifle, 2022-09-19:
This seems to be fixed in the 0.7.5 release as far as I can tell - at least the examples I reported initially

closed 2022-09-19 by itsTrifle