Mazamars312/Analogue_Pocket_Neogeo #15: Control options unavailable for Alpha 0.6.0

VideogameScrapbook, 2022-08-15:
I'm able to run Alpha 0.6.0 for games well, but can't adjust the controls. According to the directions:
"Press the home button then goto "/Settings/Cores/Player Type 1" and selectet between the 3 types"

However, I don't see a "Cores" menu when doing this. I just see the following items in the top-level Settings:
- Pocket
- Analogue OS
- Library (disabled)
- Memories
- About

Is there a prerequisite that should be added to the directions to see the Cores menu?

mbaran5, 2022-08-15:
my understanding is that this is waiting for an update from Analogue to be enabled.

esmith13, 2022-08-22:
same (expected?) issue for me. following in case this is updated once available/fixed.

DrAvedi, 2022-09-01:
I believe this is now implemented as of the latest Pocket OS (v1.1-beta-2 [2022-08-23])

It should appear as described in the ReadMe file for this core

VideogameScrapbook, 2022-09-02:
Confirmed with Analogue OS v1.1 beta 2 that this works now with newly available Core Settings:
* Player Type 1 = SNES layout (default)
* Player Type 2 = MVS mini
* Player Type 3 = Neo Geo CD

closed 2022-09-02 by VideogameScrapbook