Mazamars312/Analogue_Pocket_Neogeo #23: Analogue OS v1.1 beta 2

mddifilippo89, 2022-08-23:
[Request] Add compatibility for the new beta.

JonAbrams, 2022-08-23:
It works for me with uni-bios_1_0.rom

mddifilippo89, 2022-08-23:
Baseball Stars 2? I haven't got a single game working since the update. With uni bios 1.0 or 4.0

mddifilippo89, 2022-08-23:
Going back to bios 1.0, USA (CONSOLE) worked for me.

-------

The root issue is using bios 4.0, USA (Arcade)

mddifilippo89, 2022-08-24:
Workaround for Bios 4.0 - dip switch 0 in core options (analogue pocket menu after booting game and screen goes black)

VideogameScrapbook, 2022-09-02:
I tried a Unibios 1.0 (I believe MVS though) and it doesn't work. What's the checksum of the AES BIOS you are using?

And to developers: I suggest authors of Analogue Pocket cores always share checksum of recommended BIOS to allow for consistency when testing.

VideogameScrapbook, 2022-09-02:
I was able to get Unibios 1.0 working (CRC-32: 0ce453a0) without setting the dipswitch. Press A+B+C as soon as you see unibios splash screen to change its settings. Then use American console for settings. If you want to use a different region, you can set dipswitch 0 as explained earlier, but you have to do it every time, unlike the American console option which doesn't require going into menu every time after first setup.

JackWitherell, 2022-09-06:
I know someone will come here looking for fixes/short term workarounds.
[image]

Mazamars312, 2022-09-18:
I have removed the Dipswitch in version 0.7.5 for now to help resolve this issue. I will look into this more when I get the time in 0.8.0

closed 2022-09-22 by Mazamars312