Mazamars312/Analogue_Pocket_Neogeo #25: Option to rotate screen

sreuter, 2022-09-16:
We're working on a vertical/TATE shmup for the Neo Geo. Would adding a rotate screen feature be generally possible on the Analogue Pocket by any chance?

Mazamars312, 2022-09-18:
Should be no issue to build. As this will be done in the Analogue scaller. So advise when once you have a build and I can build a json for this

Mazamars312, 2022-09-26:
This will be added very shortly. I have this working in the current build Im working on at this moment

sreuter, 2022-09-27:
Awesome, thanks a lot! The game is still in it’s final stages of development but I’ll try it with our latest build and provide feedback as soon as it’s available.

sreuter, 2022-09-27:
I just figured that you could simply use the very early demo we published back then. https://trello.com/c/uhXMVXog/10-darksoft-multi-mister-download-attached

It has both, TATE (default on MVS) & YOKO (default on AES) in there, and you can switch it in the title screen by pressing the D button.

You could also add a XML for the game and include that in your build if you wanna, as I've seen other demos/homebrew being added as well in the past to the list.

Mazamars312, 2023-01-14:
Made the JSON and the rotation stuff here on the first video slot.

I made a directory called pneon as well. but if you have another name for it, just change it in the json

[Project Neon.txt](https://github.com/Mazamars312/Analogue_Pocket_Neogeo/files/10418439/Project.Neon.txt)
[video.txt](https://github.com/Mazamars312/Analogue_Pocket_Neogeo/files/10418443/video.txt)

closed 2023-01-14 by Mazamars312