Mazamars312/Analogue_Pocket_Neogeo #28: Distorted audio in "The Super Spy"

naveedster, 2022-09-19:
As soon as the user loads up The Super Spy, distorted audio is heard throughout the intro and in gameplay.

naveedster, 2022-09-19:
https://user-images.githubusercontent.com/80052032/190948046-1088cda1-fcad-40d6-bd50-1394e7138791.MOV

strangely turning the system off and playing it again fixed this issue! Still it is a bug.

naveedster, 2022-09-19:
There is also still popping and cracking audio for this game.

naveedster, 2022-09-19:
Here is what it sounded like the first few times I tried to play it: https://youtube.com/shorts/fD2NgV1LrEY?feature=share

Mazamars312, 2022-09-26:
Ill look into this once I get more of the Graphics sorted on this core first

Krathoon, 2022-12-14:
This also happens in Blazing Star and Metal Slug 3. It is happening all over. There is an audio bug in there.

Mazamars312, 2023-01-03:
Found the issue to be a masking issue. working on this and a few other things at this moment

Mazamars312, 2023-01-14:
Resolved in 0.8.0, this was due to a masking and offset issue with the VROMS

closed 2023-01-14 by Mazamars312