Mazamars312/Analogue_Pocket_Neogeo #33: Slow Downs on Street Hoop with V0.7.5

alesgarci, 2022-10-12:
With V0.6.0 there is no slow down in the game. Once i updated to v0.7.5 in order to be compatible with new analogue pocket firmware, heavy slow downs in this game.

Thank you very much for your effort and this neo geo core for the analogue pocket

Mazamars312, 2022-10-30:
Yes, Ive had to re-write the whole memory controllers. The next build I will be looking at speeding this up.

timotheeg, 2023-01-07:
I just received my Pocket and I see some slow downs in the various metal slugs too (on 0.7.5), and I was wondering if the Pocket just wasn't powerful enough to run the NeoGeo 😢.  But so we can expect speed improvements in the future?

davewongillies, 2023-01-07:
> I just received my Pocket and I see some slow downs in the various metal slugs too (on 0.7.5), and I was wondering if the Pocket just wasn't powerful enough to run the NeoGeo cry. But so we can expect speed improvements in the future?

https://twitter.com/UFp64/status/1610764434464739328

timotheeg, 2023-01-07:
Thanks for that link @davewongillies 🙏

Mazamars312, 2023-01-11:
Slowdowns have been reduced a lot in version 0.8.0 being release this weekend

closed 2023-01-14 by Mazamars312