Mazamars312/Analogue_Pocket_Neogeo #34: Reset button

Princepicklepants, 2022-10-27:
Core is amazing whole family is having the best time playing through these great arcade games , is it possible to move the reset game button to a menu option. My nephews are playing through the  metal slugs and keep hitting the shoulder button equivalent on the 8bitdo arcade stick and resetting the game. I know it’s user error and most people won’t do that but it would make a big difference to these young guys!

Mazamars312, 2022-10-30:
As the Neogeo core is in beta I have keep this on for now. Will look at making a option to disable this.

Princepicklepants, 2022-10-30:
That would be great thank you! Looking forward to the next update

Wabbajackw, 2022-11-07:
Same here, after a long session i reset by accident in final mission...

tarikalmashni, 2022-11-23:
> is it possible to move the reset game button to a menu option.
I literally came here just to ask the same. But I'm glad it's already covered here and the dev is already (eventually) planning to do it after leaving beta. (I'd still love to have this earlier though honestly)

closed 2023-01-11 by Mazamars312

Princepicklepants, 2023-01-11:
Awesome thanks!