Mazamars312/Analogue_Pocket_Neogeo #36: Scrambled graphics on some layers with v0.7.5

EvilRyuSp, 2022-11-16:
[image]
[image]

With version 0.7.5 I see lots of scrambled graphics in some layers. Please see screenshots for details.

This happens with all games, multiple bios files, different compilations of the core, version 0.7, different versions of the games, etc. I have reported this on Reddit and got one response of a user having the same issues. My Analogue Pocket is one of the newer ones for Group B preorders (maybe slight differences in memory tolerances in some of the new models?)

My files have been tested on a second Analogue Pocket that did not have any issues to confirm that the configuration was correct.

terminator2k2, 2022-11-16:
problem seemed to have been solved with a recompile

jere08200, 2022-12-01:
Excuse me but I have the same problem ! And I don't understand your answer. What's SEED 1 ? How to solve this problem ?

terminator2k2, 2022-12-01:
> Excuse me but I have the same problem ! And I don't understand your answer. What's SEED 1 ? How to solve this problem ?

go here, and download the core:

https://www.reddit.com/r/AnaloguePocket/comments/ywfgqg/comment/iy6iyr3/?context=3

jere08200, 2022-12-01:
thank you very much i will try that

jere08200, 2022-12-01:
excuse me but where should I put the core? in the neo geo alpha 0.7.5 common file?

terminator2k2, 2022-12-01:
in Cores\Mazamars312.NeoGeo

jere08200, 2022-12-01:
Thanks a lot ! it works now

dji85, 2022-12-02:
I got some scrambled sprites on World Heroes 2

evanphx, 2022-12-18:
Sounds like it's something up with a newer patch of analogue pockets and their memory timing. The rom posted at the link before isn't available anymore, anyone still have it to share?

terminator2k2, 2022-12-18:
> Sounds like it's something up with a newer patch of analogue pockets and their memory timing. The rom posted at the link before isn't available anymore, anyone still have it to share?

here you go:

https://easyupload.io/ijx3t3

evanphx, 2022-12-19:
@terminator2k2 Thank you!

Mazamars312, 2023-01-11:
Most of this stuff is fixed up in the new build. Should be released soon. The only game I know in the new build that has a issue is The Super Spy. which ill look at later

Mazamars312, 2023-01-17:
This should be resolved in version 0.8.1. Can you please confirm?

EvilRyuSp, 2023-01-17:
I updated to 0.8.0 yesterday and did not encounter this issue, looks good to me!

closed 2023-01-17 by Mazamars312