Mazamars312/Analogue_Pocket_Neogeo #38: Blues Journey distorted sound......(real bad)

Wesleyrpg, 2022-11-18:
Just wanted to let you know that Blues Journey has really badly distorted sound on certain sounds, and sound effects

Mazamars312, 2023-01-03:
Found the issue to be a masking issue. working on this and a few other things at this moment

Mazamars312, 2023-01-17:
This should be fixed in 0.8.1 can you please confirm

Wesleyrpg, 2023-01-17:
It was 99% fixed on 0.8.0 with only the slightest pop once in a while......i'll test 0.8.1 tomorrow for you!

Mazamars312, 2023-01-17:
0.8.1 is more on some masking for graphics, so that's a good sign

Wesleyrpg, 2023-01-18:
yes, blues journey works well now on the pocket itself.

I'll test with the dock later!

closed 2023-01-25 by Mazamars312

Wesleyrpg, 2023-01-25:
Didn't get a chance to test it on the dock, but i'm guessing it's fine.....you're welcome to close this!