Mazamars312/Analogue_Pocket_Neogeo #47: Last Hope scrambled graphics

Kypck, 2023-01-16:
Unfortunately I am not able to get this game to run in a visibly good way. the graphics are always scrambled.
Gameplay/sound seem fine. As the game works good on the MiSTer I would like to know how to assemble the rom?

Do you need to concat the C-Roms out of order?
[image]

terminator2k2, 2023-01-16:
this game ran ok on 0.7.5

[image]
[image]

but has text glitches on 0.8.0

[image]
[image]

Mazamars312, 2023-01-16:
Found the issue. Most games have an SROM size of 128K, where Last hope has a 64K Rom. 
So I didn't have an edge-case for this.

Will add this to the next update. Or you can double the SROM data twice for the temporary moment.

Kypck, 2023-01-17:
Works like a charm for the text layer. Thank you very much! 
@terminator2k2 : Any Idea why the graphics are still scrambled on my end? Used 
cat ngdt-300-c1.bin  ngdt-300-c2.bin  ngdt-300-c3.bin  ngdt-300-c4.bin > crom0
to create the crom0

terminator2k2, 2023-01-17:
use this python script :
https://www.arcade-projects.com/threads/how-to-convert-files-to-work-with-neogeo-multi.24393/
so do c1.bin + c2.bin then c3.bin + c4.bin then concatenate the 2 files together

Kypck, 2023-01-17:
It just works if you do it right :-)
Thank you very much. I searched for such information (especially knowing about Darksofts work) but did not find any infos. Really appreciate that!

Mazamars312, 2023-01-17:
Have fixed the SROM mask in0.8.1 and if you used the full crom you should be all good

closed 2023-01-17 by Mazamars312