Mazamars312/Analogue_Pocket_Neogeo #50: Crossed Swords 2 (CD Conversion) - Glitched graphics and missing lauers

GumbyXGames, 2023-01-18:
When playing the NeoGeo conversion of Crossed Swords 2, some sprite layers will flash or disappear completely. The game does not crash even if the video is completely missing or garbled.

Secondary issue is the the game can only be played with the Overdrive core. The standard core errors out with "File ID [257] too large".

GumbyXGames, 2023-01-18:
Second image of issue
[image]

mbaran5, 2023-01-18:
Error 257 is a result of having an old save file. See here: https://github.com/Mazamars312/Analogue_Pocket_Neogeo/blob/cf1c4c60ce52574a924ec8db14680d567e75d3b3/README.md?plain=1#L39

Mazamars312, 2023-01-18:
Yep the 257 error is the saves
As to the graphic bugs, which version of the neogeo core are you using? as more was fixed in 0.8.1

GumbyXGames, 2023-01-18:
> Yep the 257 error is the saves As to the graphic bugs, which version of the neogeo core are you using? as more was fixed in 0.8.1

I'm using 0.8.1. Maybe it's because I used the Overdrive core? Let me delete that save file I didn't know I had and I'll see if it runs better under the normal core.

GumbyXGames, 2023-01-18:
Deleting the save got the game working with the regular core. With that core there are no graphical issues. Closing issue. Thanks for the help

closed 2023-01-18 by GumbyXGames