Mazamars312/Analogue_Pocket_Neogeo #52: Unibios menu covers screen in garbage

nach-code, 2023-01-21:
Bringing up the Unibios covers the screen in garbage, which doesn't seem to go away after closing the unibios menu. Games are close to unplayable after using the unibios menu.

I don't know if this is in every game, as I only tested the various Metal Slug games.

>  Nope not in the core, but if you use unibios you can press start and select to pause in software

_Originally posted by @Mazamars312 in https://github.com/Mazamars312/Analogue_Pocket_Neogeo/issues/49#issuecomment-1386103186_

Mazamars312, 2023-01-25:
Have not been able to emulate this bug in 0.8.1 in all of the metal slug games I have with Unibios 4.

Have you done a clean install and updated to 0.8.1?

nach-code, 2023-01-25:
Oh I didn't notice 0.8.1 got released. After upgrading to this version, this bug is now fixed. Thank you!

closed 2023-01-26 by Mazamars312