Mazamars312/Analogue_Pocket_Neogeo #53: Question: is an overclocking option possible?

timotheeg, 2023-01-25:
Metal Slug 2, even the turbo rom, really struggles, I'm wondering whether the Pocket has enough juice to have an option to  overclock the neogeo core, so it can play on the pocket even better than the original?

I'm asking candidly because I have no idea what it takes to make it happen, or if it's possible at all.

Thanks!

Mazamars312, 2023-01-25:
First off, have you updated to version 0.8.1 for this core? 
a lot of the slowdown was resolved except for the bad coding in Metal Slug 2. I cannot help with that, but the testers I have that helped me with the core testing have confirmed that this core is much more to the normal speed of the OG Neogeo now
There is an overdrive core as well where I have overclocked the core to 16mhz (You have to enable this, and however, there are bugs that are introduced to this overdrive core) https://github.com/Mazamars312/Analogue_Pocket_Neogeo_Overdrive

closed 2023-01-25 by Mazamars312

timotheeg, 2023-01-25:
Thanks for the quick reply! I played on 0.8.0 and it's good!

I was indeed asking for something to overcome the bad coding of metal slug 2 itself, so thank you for pointing me to the overdrive repo! 🙏  I will take a look asap! 😄

timotheeg, 2024-08-17:
Hey @Mazamars312, I realized I didn't add a reply to thank you for the overdrive core

Playing Metal Slug 2 on it with no slow downs is just fantastic. Thank you! Thank you very much for your hard work on that! 🙏