Mazamars312/Analogue_Pocket_Neogeo #54: File size error

DrAvedi, 2023-01-31:
Been enjoying this core immensely since it's release.

Since updating to 8.1, I'm getting an error when loading a significant number of these Darksoft roms. "Error in framework File ID [257] too large"

Certain roms boot without issue, no error message.

I'm getting the error in too many games to list them all here. A few examples:

2020 Super Baseball
Aero Fighters 2
Fatal Fury
Ghostlop
KoF '94
Mutation Nation

Please help!!

DrAvedi, 2023-01-31:
oh, i'm using the recommended bios, uni-bios 4.

Thanks

mbaran5, 2023-01-31:
This is covered in readme. Delete any old saves that are larger than 8kb. 

https://github.com/Mazamars312/Analogue_Pocket_Neogeo#installation-and-usage

DrAvedi, 2023-01-31:
Oh, thanks. I figured it was my fault for missing some important detail.

closed 2023-01-31 by DrAvedi