Mazamars312/Analogue_Pocket_Neogeo #55: Big golf ball sprite appears next to the golfer in Neo Turf Masters

c0de90e7, 2023-02-07:
Hi,

In Neo Turf Masters ( golf game ) a big golf ball appears on the grass on the left of the golfer. Seems like a scaled or bigger version ( like 7X by 7X the size of the ball the golfer hits ) of the ball sprite.

Thank you for your amazong work.

c0de90e7, 2023-02-16:
I apologize for habing opened this; strange ad it may seem this is actually normal or present in the mame emulated game too, I looked for an original board video but found none. But 99% sure this is not an actual glitch now.

closed 2023-02-16 by c0de90e7

aphexian, 2023-09-16:
Sorry, but I have to laugh. That's what a tee box generally looks like on a golf course - the area between the two markers is where you tee off from.

[image]