Mazamars312/Analogue_Pocket_Neogeo #56: Data slot allocation size too small for homebrew/demos (ex: Bad Apple)

YLFAndy, 2023-04-14:
The current 64MB memory slot allocation is too small for loading 128MB demos (ex: bad apple) and there is no support for CROM1.

Frequency: 100%
Impact: Minor

Description:
Some demo/homebrew NeoGeo applications require > 64MB of space. Usually they include a secondary CROM file (CROM1).

Repro Steps:
Download the Bad Apple demo in zip format (available on your favorite search engine)
Create folder Assets/ng/common/bada
Copy the contents of the ZIP file into the folder
Rename the files as follows:
bada-c1.bin -> crom0
bada-c2.bin -> crom1 
bada-m1.bin -> m1rom
bada-p1.bin -> prom0
bada-s1.bin -> srom
bada-v1.bin ->vroma0
(this is simply to keep in convention with the standard json format).
Create a JSON file under Asssets/ng/Mazamars312.NeoGeo called "Bad Apple Demo.json"

Result:
Only the first character rom loads. Demo plays with corrupt graphics.

Attempted workarounds:
Adding a slot ID 10 to data.json that mirrors slot 5, but required = false, adding slot 10 to asset JSON file.
Concat crom0 and crom1 (too large, if data slot size is updated to 128MB in data.json, it loads with corrupted graphics). 

Again, super minor issue and possibly not worth fixing, but it's a popular demo and other NG homebrew software may also use the larger split CROMs. 

Playing with the CMC, SMA, and PVC values did yield some different corruptions, so perhaps there's a combo of memory mapping flags that could load a concatenated 128MB CROM0, but a cursory glance makes me think the memory is masked down to 64MB on that slot.

YLFAndy, 2023-04-14:
Seems like this was already bugged and closed. :/ Closing dupe.

closed 2023-04-14 by YLFAndy

Mazamars312, 2023-04-14:
Yep, On 64Mbyte is the higher limit for CROMs as the SDRAM in the pocket is only 64Mbytes. Sorry no more room for this.