Mazamars312/Analogue_Pocket_Neogeo #58: Pad 2 Support

jakekeebler, 2023-07-16:
Add the option for Pad 2 to remap their controls when the Analogue Pocket is docked. Pad 2 uses the default mappings currently.

Hazelwood77, 2023-12-24:
I have the same issue. Is this an OS issue, or an issue with the core? The issue is with the Neo Geo 8bit controller and the swapping of the A/B button on P2. Hopefully, there will be a fix soon but I can't complain as this is all done for free and I couldn't be more grateful!

jakekeebler, 2023-12-24:
To my knowledge, it's an OS issue. openFPGA cores are unable to add P2-P4 remapping with the current OS. Although, it would be nice to know how to change the default controls via the core's files.

Hazelwood77, 2023-12-24:
Jarred,Thank you for your response. I asked Analogue to fix the bug. If enough people complain, maybe they will do it. My second question was the default cores and the files. You answered that question as well so thank you! I appreciate all the hard work you put into the Neo geo core. I absolutely love it and hope you update it sometime in the future. I understand it’s a labor of love and we all have lives.Thank youChris Sent from my iPhone

ericmorand, 2023-12-24:
@Hazelwood77 where can we jump in and ask analog to fix the bug?