Mazamars312/Analogue_Pocket_Neogeo #64: Error in framework / Missing 'SROM' ID [3]

giacecco, 2023-09-24:
Using release 0.8.1 and Metal Slug and Metal Slug 2 for testing. I've followed the instructions at https://github.com/Mazamars312/Analogue_Pocket_Neogeo#readme for the missing files and I believe everything is in the correct folder. Both games seems to load but then I get the error in the title of this issue. Thanks!

giacecco, 2023-09-24:
The issue was the format of the games' ROMs. A game suitable for this should look like a collection of these files:

```
vroma0
crom0
fpga
m1rom
prom
srom
```

If it looks different, there's probably an issue.

closed 2023-09-24 by giacecco

ericmorand, 2023-12-16:
I have this exact issue with Metal Slug 3 and the documentation is of no help. I'm sure that I'm not putting the rom files into the right folder - or maybe eventhe rom files are not in the expected format - but without a clear documentation, I have no idea how to fix the issue.

giacecco, 2023-12-16:
Hi @ericmorand , the only way I found to make it work again is setting it up from scratch. This time I used https://github.com/mattpannella/pocket-updater-utility to install everything.

ericmorand, 2023-12-16:
Thanks. Actually, my issue was the ROMs themselves, not in the correct format. I ended up download darksoft's ROMs and they work perfectly.