Mazamars312/Analogue_Pocket_Neogeo #66: Truncated sprites in Blazing Star

TheOldCamper, 2024-02-02:
I experience occasional truncated sprites and sprite flicker mostly on boss fights
[image]

Hardware: Analogue Pocket, white (from the Preorder Fulfillment Group C: 2023)
Core: Neo Geo 0.8.1 by Mazamars312
BIOS: uni_bios_4_0
Region: Europe
Mode: Arcade
MVS/AES: Checked (Enabled)

Mazamars312, 2024-02-06:
Wonder what it could be..... The only thing I can think is that it is a bandwidth issue. unless this happens on real hardware. have we checked this as I do know there is some spritetruncation if a lot is happening on screen.

Could you confirm?

Mazamars312, 2024-02-06:
https://wiki.neogeodev.org/index.php?title=Sprites_per_line_limit

This holds some information on what could be happening.

TheOldCamper, 2024-02-07:
Yes thank you, 
Apparently it's happening on real hardware - I never experienced Blazing Star outside emulation so I wasn't expecting this!
Case closed

closed 2024-02-07 by TheOldCamper